prusa3d/PrusaSlicer-settings-prusa-fff #5: Need profiles for MK4/MMU3 with nozzle sizes other than 0.40

thess, 2025-01-25:
Posted this in the wrong repo I guess -- No response from Prusa Research. If there is no intention to support alternate nozzle sizes for an MK4/MMU3 setup, it should be stated up-front before one invests in it.

The sound of crickets is deafening.

Forum:
https://forum.prusa3d.com/forum/original-prusa-i3-mmu3-general-discussion-announcements-and-releases/support-for-additional-nozzle-sizes-on-mk4-mmu3/

On Github:
https://github.com/prusa3d/PrusaSlicer-settings/issues/265#issue-2451230010
-- and here --
https://github.com/prusa3d/PrusaSlicer/issues/12634#issue-2256956172

Vonyl, 2025-01-28:
Yes, we're not asking for support for any and all sorts of random setups. These are your own products. Please fully support them :)

JakobGlieder, 2025-02-02:
Exactly!