prusa3d/PrusaSlicer-settings-prusa-fff #6: Non-MMU filament profiles are sometimes accidentally considered compatible with MMU3 on MK4S and MK3.9S.

awpr, 2025-01-27:
### Description of the bug

The non-MMU filament profiles use the following kind of subexpression to declare themselves incompatible with MMUs: `!(printer_notes=~/.*PRINTER_VENDOR_PRUSA3D.*/ and ... and single_extruder_multi_material)`.  This appears many times throughout the profiles .ini, but for example, https://github.com/prusa3d/PrusaSlicer/blob/5c7888b11453eb96a95b49dfd4cbd3b56db64eb8/resources/profiles/PrusaResearch.ini#L7255C36-L7255C79.

[image]

Meanwhile, the MK4S-derived printer profiles do not include PRINTER_VENDOR_PRUSA3D in their printer_notes: e.g. https://github.com/prusa3d/PrusaSlicer/blob/5c7888b11453eb96a95b49dfd4cbd3b56db64eb8/resources/profiles/PrusaResearch.ini#L28259 replaces the printer_notes field and does not include that tag.  This also occurs several times throughout the printer profiles (several of the MK3.9S / MK4S variants overwrite printer_notes independently).

[image]

As a result, some user-customized filament profiles derived from the stock ones on non-MMU printers can be shown as compatible with MMU printers, despite having unsuitable default settings for the multi-material filament change parameters.  When using these profiles in MMU prints, the wipe tower tends to accumulate huge blobs that lead to collisions and layer shifts.  This symptom is reported several times on forums and bug reports, and often attributed to misconfiguration and often solved by recreating filament profiles from generic ones.  My suspicion is that most or all instances of this are caused by people carrying forward non-MMU filament profiles when installing an MMU, which _should_ be prevented by the filament profile compatibility settings.

A few examples of similar symptoms (although not clear if they're caused by filament profile carry-over):

* https://github.com/prusa3d/PrusaSlicer/issues/8086
* https://forum.prusa3d.com/forum/original-prusa-i3-mk3s-mk3-hardware-firmware-and-software-help/large-blob-on-wipe-tower/
* https://forum.prusa3d.com/forum/original-prusa-i3-mmu3-hardware-firmware-and-software-help/mmu3-mk4-wipe-tower-issuestoo-much-purged/
* https://www.reddit.com/r/prusa3d/comments/1cuxy1d/new_mmu3_with_mk4_purge_tower_blobs/

I believe this can be fixed most easily by changing the MK3.9S- and MK4S-based printer profiles to include the PRINTER_VENDOR_PRUSA3D tag, so that they're correctly detected as MMU printers and considered incompatible with non-MMU filament profiles.

### Project file & How to reproduce

[1_body_r.zip](https://github.com/user-attachments/files/18551308/1_body_r.zip)

Extruder 5 contains the problematic carried-forward filament profile that's incorrectly considered compatible with the printer; extruders 1-4 contain a working one newly created from Generic PLA on this printer.

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

Version 2.9.0

### Operating system

Windows 10 22H2 19045.5371

### Printer model

Prusa MK3.9S + MMU3

rtyr, 2025-01-27:
I would say this issue can happen only with some old custom filament profiles (from pre-MK4 time).

>  This appears many times throughout the profiles .ini

But there is also `and printer_notes!~/.*PG.*/` at the end, which makes it incompatible with all MK4/MK3.9/XL printers.

The `PRINTER_VENDOR_PRUSA3D` tag is deprecated and we will likely remove it completely, because configuration bundles are separated by printer vendor automatically for years.

awpr, 2025-02-02:
This was indeed a pre-MK4 filament profile.  Thanks for pointing out how up-to-date profiles exclude the new generation of printers.  Evidently there are still profiles cloned from pre-MK4 times out there (and probably will be as long as MK3 printers are widely used). To me that seems like a good reason to keep the PRINTER_VENDOR_PRUSA3D tag in the printer notes for another generation or two (but not rely on it in new filament profiles), to ensure the new printer profiles interact well with the filament profiles they're likely to encounter in the wild.

Ultimately this doesn't directly cause any problems other than confusion, and the fix/workaround is easy enough as long as the end-user can discover it (re-customize the filament profile and avoid using the old one), so it's not a particularly big deal.  But it'd be nice if it didn't show up as compatible in the first place.