prusa3d/PrusaSlicer-settings-prusa-fff #8: Missing Prusament PVB high-flow profiles

ddavidebor, 2025-02-13:


rtyr, 2025-02-13:
Some filament profiles are shared between HF and non-HF, thats intentional.

closed 2025-02-13 by rtyr

ddavidebor, 2025-02-14:
Probably this one is not intentional, I checked with Eono PVB filament and I can easily print it at 35+ mm3/s. Current profiles greatly limit speed.

rtyr, 2025-02-14:
It's not that simple. Some filaments are better to print slower (like fiber filled ones, silk/blends, flexibles etc.). They are slowed down even with standard nozzle.

In case of PVB the main reasons were infill quality (it does not like fast crossing infill, similar to petg), the main purpose of the filament and cooling. Since we have new parameter for crossing infill speed, we may revisit it at some point.

ddavidebor, 2025-02-14:
Thank you, very informative.