prusa3d/PrusaSlicer-settings-prusa-fff #10: Spectrum PCTG profile does not work at all

martian45, 2025-02-20:
### Description of the bug

Build in Prusa Slicer filament profile Spectrum PCTG does not work with MK4S. Nozzle temperature in profile (240°C) is far from manufacturer recommendation in data sheet (250 - 270°C), Density in profile does not match density in filament data sheet, prints are wrapping.
I believe that there is a mistake in the name of profile. This profile match PETG much more than PCTG.
Please can you check this? I have very good experience with other built in profiles which work excellent. Except of this Spectrum PCTG. 

### Project file & How to reproduce

[makitatrimrouterbase.3mf.zip](https://github.com/user-attachments/files/18883260/makitatrimrouterbase.3mf.zip)

### Checklist of files included above

- [x] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0

### Operating system

macOS 15.3.1

### Printer model

Prusa MK4S + HF0.4

rtyr, 2025-02-20:
Thank you for your feedback. The PCTG profile was actually provided by filament manufacturer.

> Nozzle temperature in profile (240°C) is far from manufacturer recommendation in data sheet (250 - 270°C).
Spectrum website says 240-270C.

We will check it, but maybe @bartosz-spectrumfilaments can comment on it.

martian45, 2025-02-20:
Thanks for reply and explanation. Yes, there is mismatch on manufacturer's web page as on attached screen. 
Fiberology PCTG build in profile works for me and Spectrum PCTG Premium while Spectrum PCTG doesn't. 
Waiting for @bartosz-spectrumfilaments comment.
 

[image]

rtyr, 2025-05-05:
Spectrum PCTG profile was already updated some time ago.

closed 2025-05-05 by rtyr