prusa3d/PrusaSlicer-settings-prusa-fff #11: Core one HF 0.6 + SPEED profile can generate horizontal and vertical moves the extruder can't keep up with

DavidR91, 2025-05-02:
I've noticed the HF 0.6 nozzle on the speed profile makes certain long horizontal + vertical moves at a rate that the machine can't actually keep up with (you get the characteristic 'clunk' sound and missing extrusions).

I'm doing this with Hatchbox PLA filament + the built-in profile for it (and an E3D 0.6 high flow nozzle)

[Zipped gcode for an affected part attached](https://github.com/user-attachments/files/20016984/fruiting_grill-Body_0.6n_0.4mm_PLA_COREONE_1h57m.zip). It affects the horizontal and vertical lines of the outline of the shape - but nowhere else.

I'm pretty certain my nozzle is not clogged or partially clogged - once these longer fast moves finish [on each layer] everything else extrudes and prints normally

rtyr, 2025-05-05:
Hatchbox profile was updated by configuration update (higher temperature for 0.6 and lower vol. limit).

closed 2025-05-05 by rtyr

DavidR91, 2025-05-05:
@rtyr sorry to be annoying on this - over the weekend I've been doing some testing and found that I can get the same behaviour from the speed profile from the stock 0.4 nozzle and using Prusament PLA. So the issue is actually not specific to hatchbox and 0.6 as I first suspected.

I've made a model that consistently repros this for me sliced with the speed profile (take 2 mins): https://www.printables.com/model/1285853-fast-continuous-extrusion-test

On my core one this misses extrusions, but the problem disappears completely if I drop down to the structural profile instead.

I'm open to this being a hardware issue but I cannot find the cause (I've eliminated the nozzle, the filament, I've checked and cleaned the gears as well as running cleaning filament. This is with the stock 6.3.2 firmware)  

@altaic I'm not 100% sure but it sounds like either it is slipping on the filament beyond a certain speed, or it cannot shove it into the nozzle at the required speed because it doesn't melt quickly enough?

rtyr, 2025-05-05:
You may also check filament path (side FS and long PTFE) or simply test printing with filament loaded directly to the extruder.

Anyway It seems this is the case for tech-support.